vhelin/wla-dx #633: .FOPEN not respecting -I flag?

Stewmath, 2024-06-03:
It seems like the FOPEN directive doesn't search directories included with the "-I" flag. I'm guessing this is an oversight, but if this is a bug then I could try to come up with a test case.

vhelin, 2024-06-03:
I quickly looked into phase_1.c:directive_fopen() and it looks like it doesn't use the directories included with the "-l" flag. I'll try to implement that in the next couple of days or later this week.

Fortunately I'll have four weeks' vacation in July so then I have all the time to fix some other recent issues as well as perhaps start to work on support for Capcom Cx4 chip...

vhelin, 2024-09-14:
Sorry this took so long, my day job has been taxing and having trouble with the lack of motivation... Anyway, if you could test this fix and let me know it works then we could close this issue.

Stewmath, 2024-09-16:
I totally understand, I didn't end up actually needing this feature but I do like to report bugs when I find them.

Most likely this is fixed but my project isn't building with the newest sources, which means... another bug report!

vhelin, 2024-09-16:
Please do them bug reports coming! That way I can know something is broken and can fix the bugs. Also if there's too many bug reports open that'll create enough mental pressure for me and motivate me to fix them even more :)

Stewmath, 2024-09-16:
Sorry, my bad - I was somehow using the wrong build. I have no idea where that build came from. Current sources are working though and this issue does seem to be fixed.

I will say that I've had less issues with upgrading wla-dx compared to the old days. The tests help a lot, I think. Less surprises in general.

closed 2024-09-16 by Stewmath