vhelin/wla-dx #640: Weird error with known-good code

Ramsis-SNES, 2024-11-03:
Hi,

I"d like to update [an older project](https://github.com/Ramsis-SNES/ct-anime-intro) of mine, just some housekeeping basically. ;-)

However, when trying to assemble it using the latest WLA sources (or v10.6 for that matter), I'm getting a strange error:

`wla-65816 -o ct_msu1.o ct_msu1.asm`
`ct_msu1.asm:96: ERROR: Cannot process "phd". Syntax error?`

If I comment out that line, it'll complain about another stack-related instruction later on. :-(

Feel free to try it out, the archive contains the bare minimum of files required:

[ct_msu_test.zip](https://github.com/user-attachments/files/17611084/ct_msu_test.zip)

You still need an original Chrono Trigger ROM (named ct.sfc) as a .BACKGROUND though.

Thanks for any help!

vhelin, 2024-11-05:
Thanks for the report! I'm not sure if I can dedicate time to this tomorrow, but hopefully on Wednesday!

vhelin, 2024-11-07:
I'm using the very latest sources here, v10.7a, but I don't see anything on CHANGELOG that would affect this case since v10.6...

Ramsis-SNES, 2024-11-08:
> Have you done "make clean" and then "make" for WLA DX?

Ugh, I didn't know that would make a difference. Thanks a lot, Ville, I tried that and it works now. :-)

closed 2024-11-08 by Ramsis-SNES

vhelin, 2024-11-08:
> > Have you done "make clean" and then "make" for WLA DX?
> 
> Ugh, I didn't know that would make a difference. Thanks a lot, Ville, I tried that and it works now. :-)

Great that it worked! :) If there are changes in the updated sources to instructions table(s), then one needs to do "make clean" + "make" to get working executables, otherwise there will be lots of instruction parsing related issues.

Ramsis-SNES, 2024-11-08:
Ah, that explains it. I'll definitely keep that in mind. :-)