vhelin/wla-dx #641: make semisuperfree sections accept defined constants as bank list

sircodalotkob, 2024-11-24:
For sections, we can provide most parameters as defined constants:
```
	.def MYBASE $1
	.def MYBANK $2
	.def MYSLOT $3
	.def MYORG $0400

	.section "test" base MYBASE bank MYBANK slot MYSLOT org MYORG
	.ends
```
Unfortunately, we cannot do that for the list of banks in semisuperfree sections:
```
	.def MYBANKS "1-3/4-6"

	.section "test" semisuperfree banks MYBANKS
	.ends
```
With or without quotes, we get this:
`_UNROLL_BANKS: Section "test" has malformed BANKS list (MYBANKS).`

It looks like wla-dx doesn't even try to resolve the expression after the banks keyword.

vhelin, 2024-11-24:
Very likely .SECTION parser doesn't accept definitions for BANKS there, just plain strings. Thanks for the report! I'll try to look into this early next week...

sircodalotkob, 2024-11-24:
It'd be quite handy to define bank lists in common include files so they can be changed in one place when the ROM layout changes.

vhelin, 2024-11-24:
I hear you. :) If I have time tomorrow I'll try to fix this...

sircodalotkob, 2024-11-24:
No rush! It's Sunday, isn't it? :)

vhelin, 2024-11-24:
Now Sunday 04:00am, but with the power of one additional mulled wine glass I think I was able to fix this. :) The fix might enable string definitions in other places as well where we are expecting a string, need to go through the change later...

sircodalotkob, 2024-11-24:
Only 3:20 am here... Just gave it a go and it works as expected! Thanks for tackling this one so quickly. Cheers! :)

vhelin, 2024-11-24:
Glad we could make WLA-DX a bit better. :)

closed 2024-11-24 by vhelin