vhelin/wla-dx #643: macros lack a pure literal parameter type

sircodalotkob, 2024-11-25:
Sometimes, it could be handy to have pure literal parameters in a macro that don't get evaluated or resolved at all.

Consider this macro which performs a complex calculation and then stores the result:
```
  .macro add_and_store
  move.w #(\1+\2),\3
  .endm
```
We can call it to store the result in a provided memory location:
```
  add_and_store 1,2,$00123456
  add_and_store 1,2,myvariable
```
However, we cannot call it to store the result in a register:
```
  add_and_store 1,2,d0
  add_and_store 1,2,(a6)
  add_and_store 1,2,(a6)+
  add_and_store 1,2,5(a3)
```
wla-dx tries to evaluate or resolve the given parameters and consequently gives us `FIX_REFERENCES: Reference to an unknown label "d0".` for the first example. Currently, there doesn't seem to be a way to have a wla-dx macro just output literals.

sircodalotkob, 2024-11-25:
The simple problem described above can be solved using a wla-dx function instead of a macro. But unlike macros, functions cannot be used to generate code.

vhelin, 2024-12-01:
I managed to device a crude hack that does pre-parsing this kind of substitution inside .MACROs for lines with assembly instructions, so you can do this. The hack is ugly, but at this point in WLA DX's life it doesn't hurt that much. :)

[image]
[image]

If you think the latest sources fix this issue, please close this issue. :) Also, let's wait for a bit so that Azure Pipelines can build all the builds and run the tests (locally the tests ran, though)...

vhelin, 2024-12-01:
Number/label substitutions should work as usual, but registers etc. are considered text for the parser:

[image]

sircodalotkob, 2024-12-02:
This works now:

```
  .macro add_and_store
  move.w #(\1+\2),\3
  .endm
```

But it seems that some operators don't work (anymore) if the second operand is a parameter.

```
  .macro and_and_store
  move.l #(\1 & \2),\3
  .endm

  .macro shift_and_store
  move.l #(\1 << \2),\3
  .endm
```
Those two yield `./src/68000/main.s:13: COMPUTE_STACK: AND is missing an operand.` and `./src/68000/main.s:13: COMPUTE_STACK: Shift left is missing an operand.`, respectively.

vhelin, 2024-12-02:
Thanks for the report, I found out why those broke (I didn't copy unreplaced mecro argument references to output...) - fix is coming up soon!

vhelin, 2024-12-02:
Ok, please try out the latest sources!

sircodalotkob, 2024-12-02:
Seems to work now! :-)

A couple of  `GOT  <expression>` are printed on the terminal during assembling, most probably some temporary debug output.

vhelin, 2024-12-02:
Oh crap, will delete those...

vhelin, 2024-12-02:
Deleted. :)

sircodalotkob, 2024-12-02:
Looks good now!

closed 2024-12-02 by sircodalotkob