vhelin/wla-dx #649: [Error] .IF needs immediate data.

cyraid, 2025-01-15:
```
.MACRO BeginProc
.ENDM

.MACRO EndProc ARGS procName
  .IF procName > $FFFF
    rtl
  .ELSE
    rts
  .ENDIF
.ENDM
```

The assembler errors with ".IF needs immediate data" for the first line with the `.IF`. Why can't I use an `.IF` to compare with a label? Should just be a #? I've also tried with `\1` to no avail.

Test:
```
testFunc:
BeginProc
  lda #43
EndProc testFunc
```

In ASM16 I would use `$` to get the current PC Address to automatically detect a long jump if needed.

vhelin, 2025-01-15:
If you replace the call to EndProc with what you have in EndProc it looks to work if testFunc is outside .SECTIONs or inside a FORCE .SECTION, it seems. I guess calling a .MACRO with a label doesn't replace the label with its address, need to fix that...

Not sure I can fix this today, probably later this week...

cyraid, 2025-01-16:
Thanks for the quick response! Yes, that would be awesome. Thank you. :)

vhelin, 2025-01-17:
Please close this issue if that commit fixed it (I added a couple of tests that passed, but just to be sure...)

cyraid, 2025-01-17:
Hi @vhelin ! Thanks for the quick response. Unfortunately it doesn't seem to work for me. I'm using the 65816 SNES compiler. I've posted a sample GitHub Project here: https://github.com/cyraid/snesdevtest

The project is in vscode. To run do Terminal > Run Build Task.. In the tools folder it has the latest wla-65816 and wlalink.exe (x64 release, built with visual studio 2022), but you can feel free to use your own.

The `res/font.chr` is a chr file with fonts which is used to print on the screen via tiles and builds `out/test.sfc` when it's done.

`inc/common.inc` is where those functions are, and you can test them with the constant at the top `TEST_PROC_ADDRESS`, if it's 1 it will use the `.IF procName > $FFFF`, but if it's 0 it'll just use short jumps/returns.

`inc/snes_header.inc` is where I put the header specifics.
`inc/snes.inc` is the SNES specific constants
`/main.asm` is where the main code is.

If it passes all tests, when you run the sfc the result should print:
```
Init
Yes
Yes
Yes
Yes
Yes
```

I changed how the Procedures are defined, and I have to say, I like your Assembler as it allows this kind of flexibility:

```
;------------------------------------------------------------------------------;
BeginProc &testFunc1 ; Test basic Param and return (same bank)
;------------------------------------------------------------------------------;
  Param &DataAddr, 0
;------------------------------------------------------------------------------;

  SetMode A8XY16

  ldy #0
  lda (DataAddr, s), y

EndProc
```

Thank you for doing this. :)

cyraid, 2025-01-17:
I could also most likely do this if I could get access to information from a label's section? Like, if I could get the current section information, and then the target label section, and see if a long jump (with code bank register 0 of course) would be required?

Would be a lot of work I'm sure. I can just remove the autodetecting jumps for now.
But what WOULD be awesome would be something like:

```
; or .IF secbank(procName) != secbank() -- no parameter would be current?
.IF bankbyte(procName) != bankbyte() -- no parameter would be current bank?
  jsl procName
.ELSE
  jsr procName
.ENDIF
```

Thanks for all your time looking into this!

vhelin, 2025-01-17:
Ah, I somehow thought you were using 6502 and just wondered how you'd have code above $ffff. :D I'm these days really tired the closer the Friday is...

Yeah, what I did doesn't take the bankbyte stuff into account when it turns that label into a number. I'll work more on this, if not today, then tomorrow after a good night's sleep. :)

vhelin, 2025-01-18:
It's getting late and I only could look quickly through the code you have there (will spend more time on this tomorrow if needed) - you are using WLA DX's capabilities there much more than I would in my asm project. :) 

But if CPU memory address space if 64K then on 65816 don't addresses > $FFFF come from .BASE (or BASE for .SECTION)? I'm not detecting BASE in your code...

If you test with the latest WLA DX sources, will your project compile?

I'll get back to you tomorrow...

vhelin, 2025-01-18:
> I could also most likely do this if I could get access to information from a label's section? Like, if I could get the current section information, and then the target label section, and see if a [...]

That bankbyte() to reference the current bank looks generic and reusable! If I can I'll add support for it as well. With the latest sources, I'm assuming, bankbyte(procName) might work already?

Anyway, let's keep on working this, I'm determined to get this to work.

One note: If you have code inside a .SECTION what is not FORCE or OVERWRITE then you cannot use labels inside it in .IF as WLALINK relocates those .SECTIONs and only it knows the labels' addresses and .IFs don't get passed to the linker, they are handled just by the assembler...

cyraid, 2025-01-18:
Wow, I like your enthusiasm! I'm doing this as it's not really for me, but to be used as a part of an editor/engine, and to help newcomers. There's a lot of gotchas and pitfalls to face new people I had to overcome.

Your assembled takes away a lot of the pain by assigning addresses for you automatically, and gives you power in preprocessing. There may be a lot of code there but the nice thing about that all is it doesn't use any bytes. It's like an SDK. :)

The goal is to reduce the amount of times people would accidentally type JSR to an RTL routine, and to debug that might take a lot of time and headaches.

Because I suppose many would run most of their code in one bank, then for functions that are a bit slower might be in another bank (I don't see many phk/plk) and just use long call. But what if they accidentally had an RTS with the call that was supposed to be RTL?

I've seen some assemblers have a `.PROC` which maybe handles some of that for them. It could essentially do what I'm doing with BeginProc and Param/Arg and EndProc.

Either way I'm happy you're so enthusiastic about trying to help. I advocate your assembler over others cause I like it best. :)

Such nice support too!

vhelin, 2025-01-19:
Thank you for your support, and happy to hear you are doing something that'll help newcomers!

If you try out the latest sources, then this should work inside a .MACRO:

.IF bankbyte(procName) != bankbyte()

I could also add a way to access current's .SECTIONs parameters, but need to add a caveat to that as if WLALINK for some reason relocates the .SECTION then the .SECTIONs data in the assembler phase is outdated and incorrect...

vhelin, 2025-01-19:
Oops, made a typo in the commit message thus the commit I made a couple of minutes ago doesn't show up in this conversation...

vhelin, 2025-01-19:
Maybe add something like

isinsidesection() - returns 1 if we are inside a .SECTION, 0 otherwise
getsection("priority")
getsection("offset")
getsection("alignment")
...

?

cyraid, 2025-01-20:
Hmm.. I can't get `.IF bankbyte(addr) != bankbyte()` to work? It's giving me a `STACK_CALCULATE: Cannot find label "initStr"!` on that line. Sorry for all the trouble!

The code:
```
;------------------------------------------------------------------------------;
; Writes a "message" using tile map characters.
;   offs     An offset into the NAMETABLE
;   msgAddr  The label/offset of the message data
;   len      The number of characters to write
;------------------------------------------------------------------------------;
.MACRO WriteMessage ARGS offs, msgAddr, len
;------------------------------------------------------------------------------;

  .IF bankbyte(msgAddr) != bankbyte()
    SetAndPushBankFromAddr msgAddr
  .ENDIF

  PrepareCallProc
    Arg16 msgAddr
    Arg16 (NAMETABLE_ADDR+offs)
    Arg16 len
  CallProc copyBytesToTiles

  .IF bankbyte(msgAddr) != bankbyte()
    plb
  .ENDIF

.ENDM
```

I have:
```
.MEMORYMAP
  SLOTSIZE $8000
  DEFAULTSLOT 0
  SLOT 0 $8000
.ENDME

.ROMBANKSIZE $8000              ; 32K per ROM Bank (LoROM)
.ROMBANKS 4
```

And:

```
.BANK 0 SLOT 0
.ORG 0
```

Which means, because of slot 0 beginning at $8000, the addresses (labels) should start at $8000. I even tried inside a `.SECTION` with `FORCE`.

About the `section` functions (including a nice `section("name")`) is lower priority as I wouldn't really need it right now. It's more for a later stage.

Also a quick aside (I'll open a new bug report), but the SPC700 `mov sp, x` and `mov x, sp` has an error that the 'sp' label isn't recognized.

vhelin, 2025-01-20:
Yikes, I need to expand my test cases... Anyway, I'll come back to this today evening, work day is just starting...

cyraid, 2025-01-20:
No worries my friend! You're doing awesome, thank you for putting your free time in for helping. :)

vhelin, 2025-01-20:
Hmm, at the point you call WriteMessage with initStr, has initStr been defined if you think of this lexically?

cyraid, 2025-01-21:
I thought that I could define things without being in a section and it would be in what I defined, like:

```
.MEMORYMAP
  SLOTSIZE $8000
  DEFAULTSLOT 0
  SLOT 0 $8000
.ENDME

.ROMBANKSIZE $8000              ; 32K per ROM Bank (LoROM)
.ROMBANKS 4

.BANK 0 SLOT 0                  ; Defines the ROM bank and the slot it is inserted in memory.
.ORG 0                          ; .ORG 0 is really $8000, because the slot starts at $8000
```

I thought this would basically be like defining a section with something like:
```
.SECTION "Base" BANK 0 SLOT 0
.ENDS
```

But without me having to define it, so I was having problems. So I wrapped my includes in a `SECTION` and then put my code in another `SECTION` and then it started working (after I defined the labels first). It seems like your `bankbyte(label) != bankbyte()` seems to be working after those changes!

I know it's probably not feasible to have labels defined after, and have the .MACRO's dealt with on a second pass, right? Man I'm asking a lot haha, you could probably even close this issue now if you like but.. I see so much potential with this assembler! It's already the best one out there.

Thanks for all your incredible support and help so far, you're awesome.

vhelin, 2025-01-22:
Thank you for the kind words, but making WLA DX assembler a multi pass assembler would be a lot of work, I think. Nowadays I'm just doing all kinds of hacks to make new features to work, not minding about writing totally nasty code if necessary LOL :D I started this project in the late 90's while learning how to code ANSI C at the same time and I was a total beginner, so I need to live with many decisions I did back in the day, not knowing about anything better... :)

If, what you did, works for you, then it's good. But beware of the fact that if those .SECTIONs are free floating the linker will relocate them. If there is a possibility of getting relocated into other banks or you make some strict assumptions about the labels' addresses and they eventually cross your borders in the linker phase, your code might fail.

I'm happy that I've programmed something that is useful for people around the world. And the fact that you can create like new Gameboy/SNES games with WLA DX is just awesome. :) The world can't have enough of those, me included.

I think I'll close this issue after I've added support for this kind of stuff:

...

isinsidesection() - returns 1 if we are inside a .SECTION, 0 otherwise
getsection("priority")
getsection("offset")
getsection("alignment")

vhelin, 2025-01-24:
I decided to introduce is() and get() with parameter strings to they can be extended later with other stuff, not just .SECTION related stuff. Good progress today with these, hoping to get them to master tomorrow evening or latest on Saturday...

vhelin, 2025-01-25:
All ideas what else parameters could be given to is() and get()  are welcome!

vhelin, 2025-01-25:
I'll close this issue, but if new stuff is needed to is() and/or get() then open a new issue

closed 2025-01-25 by vhelin