vhelin/wla-dx #650: [FEATURE] Something like ISOLATELOCAL but to make labels child labels inside MACRO?

cyraid, 2025-01-22:
Hi @vhelin ! More stuffs.. Must be unhappy with me by now.. lol
Given this code:

```
;------------------------------------------------------------------------------;
.MACRO KeyToSound ARGS offs, key, sound
;------------------------------------------------------------------------------;
keyToSound{\@+1}:

  IsButtonDown key
  beq @noKey

  MusPlayChannels sound

  WriteMessage offs, yesStr, #_sizeof_yesStr
  jmp @done

@noKey:

  WriteMessage offs, noStr, #_sizeof_noStr

@done:

.ENDM

main:
  KeyToSound $0020, KEY_Y, 1
```

Is there a way I can make "keyToSound1" generate `main@keyToSound1` and `main@keyToSound1@noKey` etc? :)

vhelin, 2025-01-23:
Hi!

I'll get back to this tomorrow, had to work overtime today (and late start for the work day)... Unless my brutal version your .MACRO works :)

[image]

vhelin, 2025-01-23:
I recommend you'll try it out and take a look at the generated list file (add -i to linker and assembler) to see what kind of labels are generated. Now after a good night's sleep I need to work once more for a while... :)

vhelin, 2025-01-23:
If you add one '@' more to all labels inside .MACRO, will it then work?

cyraid, 2025-01-23:
It does (as I kinda thought it would), but I was hoping there would be an option where the macro could do it for me? Because I don't know where some of the Macro's are going to be put in other people's code. ^_^

vhelin, 2025-01-23:
So what we need here is something like '+' instead of '@' that'll make the child label "next level" instead of directly giving the level with '@'s.

Don't worry, all features that are quite generic and reusable are worth implementing as other people might use them as well. :)

Let's keep this issue open, I'll come back to this later this week...

cyraid, 2025-01-23:
Yup, I use the +/- as well, but it's more for being able to see it in the debugger for others to track stuff down in the future. Thanks @vhelin , work well and make sure you get some extra rest!

vhelin, 2025-01-23:
Thanks! By my comment I meant something like

+keyToSound{\@+1}:
+noKey:

but then "done" would need to stay at the level of "noKey", perhaps

=done:

Need to think about this, and how to make the parser to be able to parse those labels correctly (and preprocessor)

cyraid, 2025-01-23:
Oh I see! Perhaps a new 'KEEPLEVEL' keyword for the `.MACRO`? Then even if you do:

```
.MACRO KeyToSound KEEPLEVEL ARGS offs, key, sound
keyToSound{@+1}:
@child:
@@child2:
.ENDM

main:
@child:
  KeyToSound $0000, KEY_Y, 0
```

Would be `main@child@keyToSound1@child@child2` ? What do you think?

vhelin, 2025-01-25:
I think you are correct in the sense that this kind of functionality is not needed outside .MACROs, and I like this simple way of using this feature in user's code. KEEPLEVEL maybe needs to be something else, like APPENDLABELS or CHILDLABELS. I'll look into this tomorrow...

vhelin, 2025-01-25:
According to my tests it should work now, but please test it, and if it works for you, close the issue. Thanks! :)

cyraid, 2025-01-26:
Wow, you're fast! I seem to be the bug finder, haha .. I'm getting an error `FIX_REFERENCES: Reference to an unknown label "main@noKey"` at the `beq @noKey`

```
;------------------------------------------------------------------------------;
.MACRO KeyToSound CHILDLABELS ARGS offs, key, channel
;------------------------------------------------------------------------------;
keyToSound{\@+1}:

  php

  SetMode A16

  lda lastJoy1
  and #key
  beq @noKey

  WriteMessage offs, yesStr, #_sizeof_yesStr
  jmp @done

@noKey:

  WriteMessage offs, noStr, #_sizeof_noStr

@done:

  plp

.ENDM

main:
; ...
@forever:
; ...
@isChanged:
; ...
  KeyToSound $0020, KEY_Y, 0
@doneInput:
  jmp @forever
```

cyraid, 2025-01-26:
Also, I'm not sure if this is related, but:

```
song0:

@instrumentCount = 0
@sampleCount = 1

@samples:

@@sample00:
  .INCBIN "music/song0/tada.brr"
@@@end:
@@@size = (@@@end - @@sample00)
@@@loopStart = 0

@@end:
@@size = (@@end - @samples)

@sampleDir:
  .dw MUS_SAMPLE_START + (song0@samples@sample00 - song0@samples)
  .dw song0@samples@sample00@loopStart
@@end:
@@size = (@@end - song0@sampleDir)

@upload:
  MusUploadData song0@samples, MUS_SAMPLE_START, song0@samples@size
  MusUploadData song0@sampleDir, MUS_SOURCE_ADDR, song0@sampleDir@size
  rtl
```

Gives `DIRECTIVE_ERROR: "@@size" was defined for the second time.` .. Shouldn't the second `@@size` be a part of `@sampleDir`? Maybe the second `@@size` is being read as `samples@size`? When it should be `@sampleDir@size`?

vhelin, 2025-01-26:
Oh man, I'm sloppy these days, so many bugs. I think I need a long vacation... :) I'll try to fix this properly today.

vhelin, 2025-01-26:
I think I fixed the issue when you reference generated CHILDLABELS, but now you need to prefix the references with '?' so that the parser knows it's dealing with a generated CHILDLABEL and not normal labels...

.macro "hello" childlabels
childlabel: jp ?childlabel
.endm

Appending .DEFINEs, I guess that can be done as well, just need more time for that... :)

cyraid, 2025-01-26:
@vhelin wow. Thank you for going above and beyond to help me and make your assembler even better! I'll test when I get to it next which should be a bit later.

The child defines are convenience. My import tool makes those files but I would probably definitely make use of that and others might too!

cyraid, 2025-01-27:
```
;------------------------------------------------------------------------------;
.MACRO KeyToSound CHILDLABELS ARGS offs, key, channel
;------------------------------------------------------------------------------;
keyToSound{\@+1}:

  php

  SetMode A16

  lda lastJoy1
  and #key
  beq ?noKey

  SetMode A8

  lda playChannels
  ora #(1 << channel)
  sta playChannels

  lda stopChannels
  eor #(1 << channel)
  sta stopChannels

  WriteMessage offs, yesStr, #_sizeof_yesStr
  jmp ?done

?noKey:

  SetMode A8

  lda stopChannels
  ora #(1 << channel)
  sta stopChannels

  lda playChannels
  eor #(1 << channel)
  sta playChannels

  WriteMessage offs, noStr, #_sizeof_noStr

?done:

  plp

.ENDM

main:
; ...
@forever:
; ...
@isChanged:
; ...
  KeyToSound $0020, KEY_Y, 0
@doneInput:
  jmp @forever
```

With the same sort've call structure from before, it seems to be giving an error at the `ora #(1 << channel)` line saying `Unresolved reference to "main@isChanged@channel"`? Perhaps it's thinking the arg `channel` is also included in the child labels?

vhelin, 2025-01-28:
Sorry, having issues with sleeping and divericulitis right now, need to rest...

I need to debug that "channel" issue, I think I only added label hierarchy expansion on references that begin with a '?'.

Anyway, don't put '?' to label definitions, only references to them.

cyraid, 2025-01-28:
Oh no! Get some more rest my friend! Please! You're doing great but don't rush too much. Health and rest is very important, so please rest a bit more okay? Or try to! And maybe some nice food whichever you're allowed to eat. :)

vhelin, 2025-01-28:
Thanks, but I'm building up now sleep pressure by staying awake so I'll become very sleepy and can fall asleep fast. :)

Anyway, my memory failed me once again. I hope the latest sources work now better for you, I fixed that issue with "channel" there...

vhelin, 2025-01-29:
Here's an example of that '?' usage

[image]

cyraid, 2025-01-29:
Works great! Thank you again! I hope you're resting. :D