vhelin/wla-dx #651: [FEATURE] .INC to include relative directory to the source file being assembled.

cyraid, 2025-01-24:
Man, I'm keeping you busy! Sheesh eh @vhelin ? Lower priority, but, I've found that I'm including a file, but that file includes another file (that is relative to it, in the same directory). It wouldn't know what `-I <dir>` are being included, and doesn't wanna include the wrong `common.inc` for example. Could you make include a relative file with `.INC`? Maybe `.INC RELATIVE "this_is_in_the_same_dir_as_this_file.inc"`? (to not break backwards compatibility) :)

vhelin, 2025-01-24:
Yeah, you solved my issue how to spend my free time lately. 🗡 

But sounds good, I'll come back to this after the two previous issues have been solved...

vhelin, 2025-01-30:
I accidentally added RELATIVE/LATESTDIR to .INCLUDE, then found a way to add RELATIVEDIR which is what you are asking for. I hope these will fix this issue :)

[image]

vhelin, 2025-01-30:
Tomorrow I'll look into adding these to .INCBIN if it makes sense. Too tired right now to figure out if it makes any sense :)

cyraid, 2025-01-30:
Wow, works great! You've been nailing all these feature upgrades like it's a hot knife through butter! I've been working with your assembler almost every day, for many hours. So I'm using it ALL the time haha .. Last night I worked so much till I was feeling ill! (I know, not good, but I need this demo created by the end of the month).

I'm creating an SNES Music Engine (so using both 65816 and SPC700) that loads a file made from Furnace Tracker that my importer creates. It is for a company that they're creating a NES/SNES Game Editor. It's been quite the wild ride as SNES has quite a lot of quirks that I've been discovering. Thank you so much for making my job easier and easier with all these upgrades. :)

The other feature I requested will help reduce a lot of code and complexity unless I have access to a 'current label' builtin function? Like `get("lastlabel")` or `get("label.last")` (so that I can use that to combine a new label) and the current ORG, `get("org")` ? But the `.CONTENT` Macro would be a lot more elegant. lol

vhelin, 2025-01-30:
I love your energy levels and enthusiasm! :) I'm just happy to make WLA DX better for all you users.

That SNES Music Engine sounds great, I'm a big fan of SNES music!

The CONTENT feature for a .MACRO requires some planning, I don't want to do that tired and quickly as it'll be probably quite a hack or then requires a lot from the parser, need to think about it. get("label.latest") would be quite easy to implement, perhaps get("org") as well, perhaps I'll try to implement them all...

vhelin, 2025-02-13:
Used a sleep analyzer for the 1st time yesterday night, in the morning it said I have severe sleep apnea, which explains why I'm so tired these days and have memory problems. Gotta lose weight. :) And go out more often for walks...

Anyway, I'll slowly work on this issue, next I'll see if I can add "get("org"), and then try to do that .CONTENT thing, but don't expect them to appear quickly, need to rest more...

cyraid, 2025-02-13:
@vhelin Hi friend! Nice to hear from you. I don't seem to sleep well either! Maybe I should get my sleep checked out, but I do need to lose weight too, lol ..

Your sleep is more important to me than speedy updates. Take your time, I have workarounds till then! The `get("org")` and `get("label.latest")` would be ample good for now.

Take care my friend, okay?

vhelin, 2025-02-28:
Please let me know if there's an issue with org() or orga()!

I'll be laid off, probably for a while, until the slave drivers can find me a new project or I can find a better job. I'll hopefully have some free time before going to work again so I can try to solve my sleep apnea issues etc., and have more time to spend on WLA DX among other nice things in life. :)

cyraid, 2025-02-28:
Hi @vhelin ! Nice to hear from you. A break does sound nice. :) You definitely deserve it.

I haven't tried org() yet! I didn't know you snuck that in, you silly guy lol

Most important is your health and for you to keep getting better and your sleep fixed up! I got hit with some bad infection problems myself that got me bedridden for days in a row.

I'm gonna try to pursuade the company to use your assembler and switch from asm16. Would there be any way to make it compatible just switching from that? Would you know?

cyraid, 2025-03-10:
@vhelin How you feeling? (don't take this as a nudge for the asm at all, please! lol, don't even reply about the asm! :D) .. Me recovering from a sickness, and my mother ended up getting sick reminded me of you. Your sleep getting better? Everything alright? Praying for ya!

vhelin, 2025-03-12:
Hi @cyraid ! I'm slowly returning to the land of the living. I've been now 1.5 weeks laid off and have just rested, but I think I need more time to recover from the previous work assignment... At least I can now sleep as much as my body needs.

I hope you and your mom get better soon!

I'll try to do something about that .CONTENT feature soon...

cyraid, 2025-03-12:
@vhelin I'm glad you're feeling better! :) I'm also trying to fight these infections that are trying to take me down lol

The guy I'm doing some work for is using "asm16". Would there be any way to make a mode switch for "asm16" compatibility?

I'm trying to get him to switch to your awesome assembler. He's the guy who's making/made NESmaker and now doing SNES on it. They're currently using asm16 but if I'm able to get him to switch pain free, we get a huge win. :D

But for now my biggest concern is you getting better! I'm very happy to know it's working and you're sleeping better. ^_^

cyraid, 2025-03-17:
Actually you don't have to worry about the "asm16" request. lol

vhelin, 2025-09-15:
I'll close this issue as AFAIK it's been solved. If not, please reopen it. The .CONTENT thing mentioned here is another issue...

closed 2025-09-15 by vhelin