vhelin/wla-dx #652: [FEATURE] .CONTENT inside a macro?

cyraid, 2025-01-29:
This is not too high priority, but being able to insert content in a macro would unlock things like:

```
.MACRO Data ARGS label
\1:
  .IF has("content")
    .CONTENT ; Replaces this with the block that's provided when executed
  .ELSE
    .FAIL "Need content for Data"
  .ENDIF
  \1End:
  \1Size = (\1End - \1)
.ENDM

.MACRO Proc ARGS procLabel, flags
\1:
  ; Pre-Content pushing, etc.
  .IF has("content")
    .CONTENT ; Replaces this with the block that's provided when executed
  .ELSE
    .FAIL "Need Code for Procedure"
  .ENDIF
  ; Post-Content popping, etc.
  ; RTL/RTS (depending on flags)
.ENDM

Data &someDataBlock {
  .DW $0000
  .DW $0001
}

Data &someStr { .BYTE "test" }

Proc &someProc {
  Param &A, 0
  Param &B, 2

  ; Code here

}

; Use someDataBlock as a label, and it's size someDataBlockSize
```

For whenever you've had a nice break and feel nice and rested! Hope you're doing well. Sorry for all the free-time absorbing requests!

vhelin, 2026-08-16:
I'll close this issue as I think my alternative way of doing things allows doing what the initial report requested. If not, please repoen this issue or create a new one.

vhelin, 2026-08-16:
I found this: https://impactsoft.ca/2026/07/13/protos-upgraded-the-retro-2d-video-game-console/

Also: don't steel/plastic moulds cost like millions!?

closed 2026-08-16 by vhelin

cyraid, 2026-08-16:
@vhelin Yeah it could be around 100k to 200k depending, and yeah that's pricy! I might have to use a cheaper steel, but it all depends on what we're expecting for unit sales as some steel grades wear out quicker.

And yup, that's me! If you're interested in helping out, I could definitely use someone that knows how to print custom PCB's and hardware.

Drop me a line at dennis.fehr@impactsoft.ca

vhelin, 2026-08-17:
@cyraid Hi! I'm sorry, I know nothing about hardware, only about software. I have one friend who knows how to solder, but only on the very basic level, and his solderings don't always work :)

About writing games for this system - I wish I had infinite free time for all interesting projects. :) I already game projects I've postponed for years for the lack of time...

cyraid, 2026-08-18:
@vhelin haha it's all good! Your good thoughts and wishes towards the project are great too! :)