vhelin/wla-dx #657: Makefile generation/test compile no longer works

maxim-zhao, 2025-03-26:
Moving from an older build onto newer WLA DX, I found:

1. At some point, it became necessary to add `'-t` in addition to `-M` to make it not fail if a file is not found for an `.incbin` directive. (In makefile generation mode, the goal is to instruct make as to which files are missing so it can produce them, so they are expected to be missing.)
2. The latest code fails even with this flag.

Example code:

```
.rombankmap
bankstotal 1
banksize $4000
banks 1
.endro

.incbin "file_not_found.asm"
```
Run:
```
wla-z80 -t -M -MF test.makefile -o test.o test.asm
```
Get:
```
test.asm:7: FIND_FILE: Could not open "file_not_found.asm", searched in the following directories:
./ (current directory)
test.asm:7: ERROR: Couldn't parse ".incbin".
```

Adding `-c` doesn't make it continue after this error. I didn't bisect when these changes happened yet...

maxim-zhao, 2025-03-26:
The issue seems to have been introduced by #639. That intentionally made it so the `-MF` option stops enabling test mode and fake file handling. I guess I can't fight against that - but maybe it's worth clarifying in the docs that if you want it to assume files exist, generate a .o and produce the makefile as a side effect, you should use `-M -MF <filename>`, whereas if you want it to not assume files exist, not generate a .o and produce the makefile to stdout, that's `-M`.

Presumably the goal is to behave like GCC's similar flags. I see that GCC('s preprocessor) has a `-MD` flag](https://make.mad-scientist.net/papers/advanced-auto-dependency-generation/) to opt into the side-effect mode, and otherwise defaults to not producing object files if generating makefiles. So maybe extra options would clarify it?

- `-M` to enable makefile generation
- `-MF` to specify a filename rather than stdout
- `-MD` to request .o generation
- `-MG` to enable fake file handling

Then `-t` is implied if there is `-M` and no `-MG`. No matter what you do, it will break someone's existing build... but I think making it transparent is the way forward.

vhelin, 2025-04-04:
Thanks for the report! I'll try to do something about this issue this weekend...

vhelin, 2025-08-30:
I have to admit that I've been just blindly approving the pull requests (as PRs in this project are rare) regarding the makefile generation as they've sounded good on paper, but as I don't use the feature myself I don't 100% know what's going on there. Thank you for this explanation, I'm currently working on fixing this issue the way you suggest, but as I don't 100% here know what I'm doing I'll do a PR about this and if you have the time go through it, I'd appreciate it. :)

maxim-zhao, 2025-09-02:
I'll have to remember the context and then confirm.

vhelin, 2026-02-01:
Too late to test that branch today, but now I have started to work on this issue and will try to get it to work as you described asap unless it already does work like you describe.

Just define -M first and after that -MF/-MG and -MD should work.

maxim-zhao, 2026-02-09:
OK, testing...

Old makefile rule is now broken:
`$(WLAZ80) -M -t -D LANGUAGE=$* -o generated/$*/ps1jert.sms.o -I asm asm/ps1jert.sms.asm > $@`
...because `-t` no longer implies fake file handling. Amending to:
`$(WLAZ80) -M -MG -D LANGUAGE=$* -o generated/$*/ps1jert.sms.o -I asm asm/ps1jert.sms.asm > $@`
Works! :)
Amending to:
`$(WLAZ80) -M -MG -MF $@ -D LANGUAGE=$* -o generated/$*/ps1jert.sms.o -I asm asm/ps1jert.sms.asm`
Works! :)

I misunderstood about `-o output.o` - now I understand, it will generate the output file (whether explicitly configured or implied from the input file) when in makefile generation mode. I proposed that it should only do that if also passing `-MD`, to match GCC behaviour, but practically one of two things will happen:
 
1. The dependencies of this file are all present, and the .o is perfectly correct
2. One or more dependencies is not present, so the makefile rule will trigger their generation, and then they have a newer timestamp than the .o -> make will re-run the command

So it's OK for me. Maybe add something to the docs to clarify around the breaking behaviour here and suggest some examples.

closed 2026-02-09 by maxim-zhao