vhelin/wla-dx #664: Operand size mnemonic placement bug (and an idea for a new one)

stove-panini, 2025-08-09:
I'm using the HuC6280 flavor of WLA (compiled from master as of this writing). I was working with the operand size issue that a lot of folks seem to run into when using WLA for 6502-based CPUs and ran into what I think is a bug.

```asm
.16bit

sta $50    ; 8D 50 00
sta $50.b  ; 8D 50 00 (!)
sta.b $50  ; 85 50
```

Shouldn't the size mnemonic work on the operand? I think it's even recommended to place it there instead of after the instruction.

This brings me to another issue with the HuC6280 since the zero page for the PC Engine begins at $2000.

When working in `.16bit` mode, you have to do a little extra coercion when the zero page address is a symbol:
```asm
.16bit

; The PC Engine's zero page starts at $2000 :')
.ramsection "zp" slot 1 orga $2000 force
foo db
.ends

sta foo    ; 8D 00 20
sta foo.b  ; 8D 00 20
sta.b foo  ; FIX_REFERENCES: Value ($2000) of "foo" is too much to be a 8-bit value.
sta <foo   ; 8D 00 00
sta <foo.b ; 8D 00 00
sta.b <foo ; 85 00

```

What about another size mnemonic, `.z`, that forces the symbol to be interpreted as 8-bit no matter what? This way, it won't matter where the zero page is for a given system. Easier said than done, I bet! 

Anyway, thanks for all of your hard work on this project. I'm having a lot of fun w/ WLA!

vhelin, 2026-03-29:
I have a gut feeling that .16bit overrides $50.b, but not sta.b as they are parsed in different parts of the assembler...

vhelin, 2026-04-03:
The first issue should now be solved, I hope. :)

About adding support for .z, I guess it might make sense, and might also be useful outside HuC6280 also. Implementing it is a bigger task as I need to add it to the assembler and the linker, to quite a few places. But I'll try to do that next.

vhelin, 2026-04-09:
Sorry, I had to rename .z to .#b as .z could have been a struct member