vhelin/wla-dx #668: Allow labels to be used in ds/dsb

H-A-M-G-E-R, 2025-08-17:


vhelin, 2025-08-17:
You should be able to use labels already, but only "relatively"; like "labelA - labelB". This is because the linker places all free labels and .DSB needs to know the size already in the assembler.

Personally I'm not going to fix this issue as it would be just too much work, but I'll leave this issue open if someone else wants to do it. :)

vhelin, 2025-08-17:
Examples, please :)

vhelin, 2025-11-30:
Hmm, I think first you need to use .STRUCT to define a struct. Then use .DSTRUCT to create an instance of it. Like here:

```
.struct SomeStruct
  someData ds 4
.endst

.dstruct SomeStruct1 instanceof SomeStruct values
  someData: label1-label2 // you could put a label here as well, I think
.endst
```

closed 2025-12-28 by vhelin