vhelin/wla-dx #674: 65816|SNES|LOROM - Data reported outside of sections does not factor for the slot starting address

jwrayth, 2025-09-24:
WLA-DX 10.5 - Building for SNES LOROM config, SLOT 0 has a starting address of $8000.

If you have data/code declared outside of a section, it does not add the slot starting address. 

Example from output;

```
ROM bank 0 (32337 bytes (98.68%) used)
  - Outside .SECTIONs (1089 bytes)
    - Used space at $613e-$655e (1057 bytes).
    - Used space at $7fb2-$7fb5 (4 bytes).
    - Used space at $7fc0-$7fdb (28 bytes).
  - Sections (31248 bytes)
```

The actual addresses for the data is the reported values + $8000

vhelin, 2025-11-30:
Is this issue still present in the latest WLA DX sources? I mean WLA DX 10.5 is quite old...

vhelin, 2026-08-16:
What is reported there are ROM addresses, not CPU addresses. If I understand this issue correctly it is a misunderstanding of how the assembler reports the information. I'll close this issue, but if you can come up with a better/more verbose explanation what is wrong in the latest sources I'll take another look.

closed 2026-08-16 by vhelin